vadimgrn/usbip-win2 #100: usbredir support

anonymix007, 2025-01-20:
[usbredir](https://gitlab.freedesktop.org/spice/usbredir/-/blob/main/docs/usb-redirection-protocol.md) is yet another protocol for USB device redirection.
It has some advantages over USB/IP, i.e. official cross-platform server. And in general, the protocol seems to fix some of the well-known USB/IP issues.

At some point there was even a somewhat working Linux driver [submitted](https://lore.kernel.org/all/1435700650-640-1-git-send-email-jwhite@codeweavers.com/) upstream, but was it rejected and no one worked on that since.

It would be great to have usbredir support here.

vadimgrn, 2025-01-20:
I don't think it should be implemented. It seems everything is implemented in userspace. This means poor performance and big latency. The popularity of this solution is also a big question. The more important is to add USBip server implementation as KMDF driver. I know how to do that, but it requires a lot of time.

anonymix007, 2025-01-20:
It would be great if this is going to be implemented one day.
usbredir is quite popular. It's the go-to solution for USB device passthrough into a QEMU VM and is used in i.e. [kubernetes](https://github.com/kubevirt/kubevirt/tree/main/pkg/virtctl/usbredir) and SPICE.
As for the performance or latency, it's hard to say anything without tests.

closed 2025-04-13 by vadimgrn