vadimgrn/usbip-win2 #102: Some files are missing when trying to open the solution in VS 2022

ddbb2017, 2025-01-22:
I just downloaded this project and tried to open it in VS 2022 to receive this error:

`C:\path-to\usbip-win2-master\userspace\wusbip\wusbip.vcxproj : error  : The imported project "C:\path-to\usbip-win2-master\vcpkg\scripts\buildsystems\msbuild\vcpkg.props" was not found. Confirm that the expression in the Import declaration "C:\path-to\usbip-win2-master\vcpkg\scripts\buildsystems\msbuild\vcpkg.props" is correct, and that the file exists on disk.  C:\path-to\usbip-win2-master\Directory.Build.props`

vadimgrn, 2025-01-23:
https://github.com/vadimgrn/usbip-win2/tree/develop?tab=readme-ov-file#build-visual-studio-solution

ddbb2017, 2025-01-23:
You should probably mention in the requirements that one needs to download and [install Git](https://github.com/git-guides/install-git) first, and then clone this project using Git CLI before running your `bootstrap.bat`. Not everyone uses Git and may just download your project via a web browser.

```
C:\C++>git clone https://github.com/vadimgrn/usbip-win2
Cloning into 'usbip-win2'...
remote: Enumerating objects: 17483, done.
remote: Counting objects: 100% (2025/2025), done.
remote: Compressing objects: 100% (491/491), done.
remote: Total 17483 (delta 1665), reused 1550 (delta 1534), pack-reused 15458 (from 3)
Receiving objects: 100% (17483/17483), 5.84 MiB | 6.48 MiB/s, done.
Resolving deltas: 100% (13705/13705), done.

C:\C++>cd usbip-win2

C:\C++\usbip-win2>bootstrap.bat
Submodule 'vcpkg' (https://github.com/microsoft/vcpkg.git) registered for path 'vcpkg'
Cloning into 'C:/C++/usbip-win2/vcpkg'...
Submodule path 'vcpkg': checked out 'b588ddc8230705392c53448597f5892966a8db15'
Downloading https://github.com/microsoft/vcpkg-tool/releases/download/2025-01-20/vcpkg.exe -> C:\C++\usbip-win2\vcpkg\vcpkg.exe... done.
Validating signature... done.

vcpkg package management program version 2025-01-20-ca7c14e5a1929895e4124094788deb7ea0ac2e58

See LICENSE.txt for license information.
```

Also, the build process is overly complicated. It took over 15 minutes to build a Debug configuration. The build script downloaded and ran a bunch of packages and ballooned the `usbip-win2` folder to 9.4 GB. 

What are you installing/building there?

lazy5f, 2025-01-23:
By the way, it is open source and we can always check it by ourselves.

> What are you installing/building there?

vadimgrn, 2025-01-24:
> You should probably mention in the requirements that one needs to download and [install Git](https://github.com/git-guides/install-git) first, and then clone this project using Git CLI before [...]

Are you serious? Run bootstrap.bat and press F7 to build is overcomplicated? NuGet has old versions of wxWidgets that are never updated, that's why it builds ONCE.

nefarius, 2025-02-10:
> Not everyone uses Git and may just download your project via a web browser.

No. Every serious developer has Git installed, **nobody** who knows what they're doing downloads a source tarball...

nefarius, 2025-02-10:
> Also, the build process is overly complicated. It took over 15 minutes to build a Debug configuration. The build script downloaded and ran a bunch of packages and ballooned the usbip-win2 folder to 9.4 GB.

Welcome to any Software with a GUI. Building dependencies is only necessary once, subsequent builds use the cached object files. Also, you can always build individual components instead of the entire solution if you don't need the GUI apps.

closed 2025-04-13 by vadimgrn