vadimgrn/usbip-win2 #109: USB/IP Issue Description

Andreyokey, 2025-04-28:
When using USB/IP on Orange Pi Zero 3 running Ubuntu Noble (Linux 6.1.31-sun50iw9), the following issue has been identified:
When a client connects to a USB device via usbip and then disconnects improperly (e.g., when the client computer is powered off without properly detaching the device), the device remains in a "busy" state. In this situation, the usbip port command shows no active connections, however, new clients cannot access this device. The only way to free the device is to manually execute these commands:
bashsudo usbip unbind --busid=X-Y
sudo usbip bind --busid=X-Y
where X-Y is the busid of the respective device.
It appears that the usbip driver retains connection information even after the physical connection is broken, but doesn't provide convenient means for automatically releasing devices when communication with the client is lost.

vadimgrn, 2025-04-29:
This is not a bug, but feature of TCP/IP protocol. When a tcp/ip connection is terminated not gracefully, the other side cannot detect it. If TCP keepalive is used, the server will detect a disconnection after two hours (default value).

Andreyokey, 2025-04-30:
@vadimgrn Despite proper TCP keepalive configuration on the server:
net.ipv4.tcp_keepalive_time = 30
net.ipv4.tcp_keepalive_probes = 5
net.ipv4.tcp_keepalive_intvl = 5
The following behavior is observed:

The USBIP server does not automatically disconnect the device when the client connection is lost
The usbip unbind command is not automatically triggered
The device remains bound regardless of the waiting time (tested with intervals of 60, 120, and 300 seconds)

The issue appears to be related to the USBIP server component, which does not properly process signals from the TCP keepalive mechanism to release devices.
I understand that you are developing the client side, however, you might have information about how USBIP on the server side should respond to connection termination and why the automatic resource release mechanism is not working in this case.
I would greatly appreciate if you could suggest which direction I should look for information to resolve this issue.
Thank you enormously for your incredible work developing such an excellent client!

Andreyokey, 2025-04-30:
[22 minutes ago](https://github.com/vadimgrn/usbip-win2/discussions/108#discussioncomment-12997846)
I understand that the issue is most likely on the USBIP server side. Certainly all users encounter this problem, and in my opinion it's critical — the server cannot detect improperly disconnected clients, and USB devices are only released after manually executing unbind.
I could write a script to automatically disconnect devices when a client disconnects, but this only works in the case of 1 USB and 1 client. The difficulty is that when using multiple USB devices and multiple clients, it's unclear how to determine which specific client was using which specific device using usbip tools on the server.
I'm not sure if there's a simpler solution available that I'm missing. Perhaps there's a configuration setting or a different approach that would solve this more elegantly?

seastwood
[14 minutes ago](https://github.com/vadimgrn/usbip-win2/discussions/108#discussioncomment-12997911)
Yeah, I agree it's a critical problem. I made a USBIP Client Linux GUI and the way I handle this issue is by using ssh connections to automate binding/unbinding when attaching a device. Any time a device is attached through the GUI it does an unbind first, then binds, then, attaches. No need to manually touch the server side terminal. It's a work in progress, but that's I how I personally dealt with it. Perhaps the Windows client could implement something like this as well.

vadimgrn, 2025-05-02:
Linux usbip daemon does not use TCP_KEEPALIVE at all. Linux usbip client enables TCP_KEEPALIVE with default TCP_KEEPIDLE/TCP_KEEPINTVL/TCP_KEEPCNT. You can build your own usbipd that will set these socket options.

See https://github.com/torvalds/linux\tools\usb\usbip\src\usbipd.c, recv_request_import. Add usbip_net_set_keepalive(sockfd) after usbip_net_set_nodelay(sockfd).

Set keepalive options, timeout will be TCP_KEEPIDLE + TCP_KEEPCNT*TCP_KEEPINTVL seconds.

```
int opt = 30; // seconds
setsockopt(sockfd, SOL_TCP, TCP_KEEPIDLE, (void*)&opt, sizeof(opt));

opt = 9;
setsockopt(sockfd, SOL_TCP, TCP_KEEPCNT, (void*)&opt, sizeof(opt));

opt = 10;
setsockopt(sockfd, SOL_TCP, TCP_KEEPINTVL, (void*)&opt, sizeof(opt));
```

Andreyokey, 2025-05-03:
Hi @vadimgrn :)
Huge thanks for your solution :)
I got it working successfully. What I did:

Added the <netinet/tcp.h> header to usbipd.c
Added usbip_net_set_keepalive(sockfd) after the usbip_net_set_nodelay(sockfd) line
Then added the keepalive parameter settings (using IPPROTO_TCP):
int opt = 30; // seconds - time until first probe
setsockopt(sockfd, IPPROTO_TCP, TCP_KEEPIDLE, (void*)&opt, sizeof(opt));

opt = 9; // max number of probes
setsockopt(sockfd, IPPROTO_TCP, TCP_KEEPCNT, (void*)&opt, sizeof(opt));

opt = 10; // seconds between probes
setsockopt(sockfd, IPPROTO_TCP, TCP_KEEPINTVL, (void*)&opt, sizeof(opt));

The only adjustment I had to make was using IPPROTO_TCP instead of SOL_TCP since SOL_TCP wasn't defined in my system. After compiling and installing, the server now automatically releases devices when the connection to a client is lost!
Your help was invaluable! Thank you for pointing me to the solution for this critical issue 👍 👍 👍

vadimgrn, 2025-05-03:
You're welcome!

closed 2025-05-03 by vadimgrn