vadimgrn/usbip-win2 #110: USB device connects but loses connection after a few seconds, crashing Raspberry Pi 3 (model B v. 2)

lukas-licek, 2025-05-06:
I am trying to attach my Logitech Unify Receiver from my Raspberry Pi with a fresh install of 

> Linux raspberrypi 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64

to my Windows 11 PC 

> Version 24H2 (Build 26100.3775).

I have installed USBip-0.9.7.3-x64-release.exe from [ossign](https://github.com/ossign/vadimgrn--usbip-win2/releases).

On my Raspi i have executed the following commands

> sudo modprobe -a usbip-core usbip-host

> root@raspberrypi:/home/lukas# usbip list -p -l
busid=1-1.1#usbid=0424:ec00#
busid=1-1.5#usbid=046d:c52b#

> root@raspberrypi:/home/lukas# usbip bind --busid=1-1.5
usbip: info: bind device on busid 1-1.5: complete

> root@raspberrypi:/home/lukas# usbipd
usbipd: info: starting usbipd (usbip-utils 2.0)
usbipd: info: listening on 0.0.0.0:3240
usbipd: info: listening on :::3240

Then on my Windows 11 PC i executed 

> PS C:\Program Files\USBip> .\usbip.exe list -r 192.168.0.5
Exportable USB devices
======================
   1-1.5   : Logitech, Inc. : Unifying Receiver (046d:c52b)
           : /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5
           : (Defined at Interface level) (00/00/00)

resulting in the Raspi log to continue on as

> usbipd: info: connection from 192.168.0.11:50813
usbipd: info: received request: 0x8005(5)
usbipd: info: exportable devices: 1
usbipd: info: request 0x8005(5): complete

To connect the device on Windows i used

> PS C:\Program Files\USBip> .\usbip.exe attach -r 192.168.0.5 -b 1-1.5
succesfully attached to port 1

which resulted in a succesful connection according to the usbipd log on the Raspi

> usbipd: info: connection from 192.168.0.11:50825
usbipd: info: received request: 0x8003(5)
usbipd: info: found requested device: 1-1.5
usbip: info: connect: 1-1.5
usbipd: info: request 0x8003(5): complete

Unfortunately after about 5 seconds the Windows _device _disconnected_ sound plays and my Raspi completely crashes, losing the SSH connection and not reconnecting again

> client_loop: send disconnect: Connection reset

On my PC the device shows as detached

> PS C:\Program Files\USBip> .\usbip.exe detach -p 1
error: Das Gerät ist nicht angeschlossen. (_The device is not attached_)

The device does not work in the few seconds in which it is connected according to the log as well. Is there some step I am missing? I have not enabled the Windows Test Signing Mode, nor does it seem like I would need to when using the signed executable from ossign. Thanks for any help

vadimgrn, 2025-05-07:
https://github.com/vadimgrn/usbip-win2/tree/develop?tab=readme-ov-file#
Linux kernel crashdump is required too.

msmuenchen, 2026-01-04:
@lukas-licek Try a Raspberry Pi 5 as server... something with the older Pi cores may be shot kernel-side. I hit a similar bug on a Zero 2 W which IIRC uses the same CPU core and USB architecture as the Pi 3.

@vadimgrn I attached a crashdump on my bugreport to linux-usb: https://lore.kernel.org/linux-usb/CAEhB=Bt13wr1+cc1rfzwPxnEEuawMgTew=RJic2FsVb=Gccm+A@mail.gmail.com/T/#u