vadimgrn/usbip-win2 #112: Version Mismatch Between Windows Client and Linux Server Causes Connection Storm

Andreyokey, 2025-06-15:
🐛 Problem Description
The Windows USBIP client and Linux USBIP server have incompatible version numbers, causing continuous connection failures and network congestion.
🔍 Environment
Linux Server (Orange Pi Zero 3):

OS: Armbian (ARM64)
USBIP version: usbip-utils 2.0
Kernel version: Linux 6.x
Server version reported: 273 (0x0111)

Windows Client:

Windows 10/11
USBIP-WIN client
Expected version: 0 (0x0000)

⚠️ Issue Details
Error Message
usbip: error: USBIP Kernel and tool version mismatch: 0 273:
Behavior

Windows client connects to Linux server every 30 seconds
Version check fails immediately
Connection drops, client retries after 30 seconds
This creates ~120 failed TCP connections per hour
Network infrastructure (especially low-memory routers) becomes overloaded

🚨 Real-World Impact
To be honest, I'm not entirely sure if this version mismatch is the root cause, but I've noticed a correlation. Randomly, internet connectivity drops for the entire office - all devices lose internet access simultaneously. The only thing that restores connectivity is either:

Rebooting the router, OR

Router Log Evidence (ASUS RT-N10PV2, 32MB RAM):
kernel: nf_conntrack: table full, dropping packet
kernel: osl_ctfpool_add: skb alloc of len 1968 failed  
kernel: swapper: page allocation failure. order:0, mode:0x20
kernel: Normal free: 192kB min: 512kB
The router runs out of memory and starts dropping packets. This happens more frequently when USBIP is running with these version mismatch errors.

Is there a way to disable version checking during connection? Even if it's not officially supported, any configuration option or workaround would be helpful.
I understand this might not be the "proper" solution, but the current situation makes the network infrastructure unstable, and I'm trying to isolate whether these constant failed connections are contributing to the problem.

Andreyokey, 2025-06-18:
Everything works fine actually - both local network connections and external internet connections to the USBIP server are successful.
However, I just noticed something concerning: when I tried to connect from external internet using usbip win2, the entire office internet went down immediately and it looks like the router experienced buffer overflow (similar to a DOS attack effect?).
The router logs show the same memory exhaustion pattern:
kernel: nf_conntrack: table full, dropping packet
kernel: osl_ctfpool_add: skb alloc of len 1968 failed  
kernel: swapper: page allocation failure. order:0, mode:0x20
kernel: Normal free: 192kB min: 512kB

vadimgrn, 2025-06-18:
Try the other router (Asus, Netgear, etc). It seems like the issue with the router's firmware.