vadimgrn/usbip-win2 #124: GUI Attach button is disabled while the device can be attached from CLI

drajabr, 2025-08-05:
Hi,
Many thanks for your amazing work!

I'm using [dorssel/usbipd-win](https://github.com/dorssel/usbipd-win) as a server on another windows laptop to mainly pass the usb mouse and keyboard to my main PC. 

The issue is: for the keyboard I can use attach from the GUI and set auto-attach option without problem, but for the mouse, its grayed out, as it seems the mouse itself is not reporting its details or something. However, I can use the CLI command to attach the mouse manually using --bus-id, as appers in the screenshot below:

[image]

I would love if we can have override for the gui to allow attaching "invalid" devices.

vadimgrn, 2025-08-05:
Hi, I will try to reproduce this issue. I haven't had this problem with a Linux server.

vadimgrn, 2025-11-04:
This issue has fixed in master branch. I can give you wusbip.exe if you need it.

closed 2025-11-04 by vadimgrn