vadimgrn/usbip-win2 #125: Xbox Wireless Adpter not working

wes1993, 2025-08-31:
Hello,
Thank you so much for the fantastic project. 
I am trying to connect the Xbox Wireless Adapter dongle to my Linux server running Ubuntu, so that I can use it on my Windows gaming PC. I was able to configure everything, and I can connect the USB dongle to my Windows machine but the controller doesn’t seem to connect to the adapter.

[image]

[image]

Warm regards
Stefano

zhonghuihong, 2026-08-19:
[image]
The Xbox Wireless Adapter also works normally for me. The controller connects successfully, and the controller’s 3.5mm headset jack works as well. When I plug in a headset, I can hear audio normally.

tduffy83, 2026-08-20:
> The Xbox Wireless Adapter also works normally for me. The controller connects successfully, and the controller’s 3.5mm headset jack works as well. When I plug in a headset, I can hear audio normally.

Which version are you using?